TexasInstruments/edgeai-gst-plugins #27: ColorConvert: I420 content issue with some resolutions

emmadrigal, 2021-07-30:
For some of the resolutions the stride calculation appears to be incorrect, resulting in a corrupted frame at the exit, for example:

```
gst-launch-1.0 \
videotestsrc num-buffers=20 ! \
"video/x-raw,format=RGB,width=320,height=240" ! tiovxcolorconvert ! \
"video/x-raw,format=I420" !  filesink location=image.bin
```

emmadrigal, 2021-07-30:
gst-launch-1.0 \
videotestsrc num-buffers=20 ! \
"video/x-raw,format=RGB,width=1920,height=1080" ! tiovxcolorconvert ! \
"video/x-raw,format=I420" ! filesink location=image.bin

emmadrigal, 2022-02-15:
This is caused by incorrect handling by kmssink of the buffer width and height which is included as part of the GstVideoMeta in the buffer.

ravi-rahul, 2023-03-01:
this is fixed now

closed 2023-03-01 by ravi-rahul