TexasInstruments/edgeai-gst-plugins #45: tiovxmultiscaler not working with fpsdisplaysink

ravi-rahul, 2021-08-11:
trying to use below launch command to measure effective framerate of the pipeline using  fpsdisplaysink plugin:

gst-launch-1.0 videotestsrc ! video/x-raw, format=NV12, width=1280, height=720 ! tiovxmultiscaler ! video/x-raw, width=320, height=180 ! fpsdisplaysink video-sink=fakesink -v

But this pipeline dose not run

jafet-chaves, 2021-08-12:
Running the pipeline with GST_DEBUG=2 reveals that the fpsdisplaysink is reporting that the buffers are arriving without timestamp

```
0:00:00.944523990  5583     0x10cda320 WARN                   pango gstbasetextoverlay.c:2999:gst_base_text_overlay_video_chain:<fps-display-text-overlay> buffer without timestamp, discarding
```

ravi-rahul, 2021-08-23:
Verified

closed 2021-08-23 by ravi-rahul