TexasInstruments/edgeai-gst-plugins #47: application hangs when setting the pipeline to NULL with tiovxcolorconvert

ravi-rahul, 2021-08-11:
application hangs when setting the pipeline to NULL
when tiovxcolorconvert is used in place of videoconvert

I will be sharing the steps to reproduce

ravi-rahul, 2021-08-19:
steps to reproduce

clone the edge ai apps git repo under /opt
https://git.ti.com/git/edgeai/edge_ai_apps.git

Run the init script
cd /opt/edge_ai_apps
. ./init_script.sh

download models
./download_models.sh --recommended

Run python app
cd apps_python
./app_edgeai.py ../config/image_classification.yaml -nv

Try stopping the app using ctrl-C

ravi-rahul, 2021-08-19:
the issue comes randomly
need to run 3-5 times to get the issue

app hangs at 
gst_wrapper.py +143
where pipeline is being set to NULL

ravi-rahul, 2021-08-23:
The issue is still present after #45 got fixed

its more consistent in cpp apps

**steps to run cpp apps**

clone the edge ai apps git repo under /opt
https://git.ti.com/git/edgeai/edge_ai_apps.git

Run the init script
cd /opt/edge_ai_apps
. ./init_script.sh

download models
./download_models.sh --recommended

Build cpp apps
mkdir build
cd build
cmake ..
make -j2
cd ..

Run cpp app
cd apps_python
./bin/Release/app_edgeai ../configs/image_classification.yaml -nv

Try stopping the app using ctrl-C

ravi-rahul, 2021-08-23:
log when GST_DEBUG=4 is enabled

0:00:18.911624315  2163     0x1c91e600 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline0> child 'capsfilter3' changed state to 2(READY) successfully
0:00:18.911644140  2163     0x1c91e600 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<capsfilter5> current PAUSED pending VOID_PENDING, desired next READY
0:00:18.911676430  2163     0x1c91e600 INFO              GST_STATES gstelement.c:2688:gst_element_continue_state:<capsfilter5> completed state change to READY
0:00:18.911693475  2163     0x1c91e600 INFO              GST_STATES gstelement.c:2588:_priv_gst_element_state_changed:<capsfilter5> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:18.911714525  2163     0x1c91e600 INFO              GST_STATES gstbin.c:2952:gst_bin_change_state_func:<pipeline0> child 'capsfilter5' changed state to 2(READY) successfully
0:00:18.911731775  2163     0x1c91e600 INFO              GST_STATES gstbin.c:2503:gst_bin_element_set_state:<tiovxcolorconvert1> current PAUSED pending VOID_PENDING, desired next READY

It hangs here when tiovxcolorconvert1 state is changing from PAUSED to READY

rrcarlosrodriguez, 2021-08-23:
@ravi-rahul @shyam-j we found the origin of the issue. It is not an element issue but more like a pipeline configuration problem.

We tracked down the issue to the following pipeline:

```
GST_DEBUG=2,*ti*:9 gst-launch-1.0 videotestsrc pattern=ball ! "video/x-raw,width=1280,height=720,format=NV12,framerate=30/1" ! tiovxcolorconvert ! "video/x-raw,width=1280,height=720,format=NV12,framerate=30/1" ! tiovxmultiscaler name=m m. ! "video/x-raw,width=640,height=480,format=NV12,framerate=30/1" ! queue ! tiovxcolorconvert ! "video/x-raw,width=640,height=480,format=RGB,framerate=30/1" ! appsink max-buffers=2 eos=false drop=true
```

This pipeline gets locked when playing, and the root cause is on the output buffer pool of the tiovxcolorconvert. As you can see, in this pipeline the appsink has a max-buffers=2, which means the element will store only 2 buffers, and will drop the oldest as soon as new buffers arrive. However, the tiovxcolorconvert output buffer pool size is also 2, which causes a lock, because the appsink will retain 2 buffers, but those are the 2 buffers that the colorconvert can use in the pool, and the entire flow is locked.

This same issue happens in your application but in the transition to NULL, when a CTRL-C happens, the buffers' flow gets paused and locked, and the elements can't finish the element transition.

This can be fixed by increasing the output buffferpool size, for example, I tested with 4 and the CTRL-C issue never happened again. You can apply the following patch to your application:

```
diff --git a/apps_cpp/common/src/edgeai_demo_config.cpp b/apps_cpp/common/src/edgeai_demo_config.cpp
index 03382ac..8921bee 100644
--- a/apps_cpp/common/src/edgeai_demo_config.cpp
+++ b/apps_cpp/common/src/edgeai_demo_config.cpp
@@ -285,7 +285,7 @@ int32_t InputInfo::addGstPipeline(const vector<string>     &preProcStr,
             auto const &sensorPath = " " + splitStr
                                      + ". ! queue"
                                      + mosaicStr
-                                     + " ! tiovxcolorconvert"
+                                     + " ! tiovxcolorconvert out-pool-size=4 "
                                        " ! video/x-raw, format=RGB"
                                      + buffCntStr
                                      + srcElemNames[j];

```

Additionally, in your application you 2 paths, and you might wonder why this issue happens with one of the paths only? It is because the other path has a videobox in between the appsink and the tiovxcolorconvert, and that element releases the buffers fast enough, and the tiovx pool is not shared between the colorconvert and the appsink.

Please test above patch and let us know if that fixes this issue

ravi-rahul, 2021-08-24:
Thanks @rrcarlosrodriguez ,

The issue is fixed after changing the out buffer pool size to 4
and thanks for the detailed explanation, it is very helpful

closed 2021-08-24 by ravi-rahul