TexasInstruments/edgeai-gst-plugins #49: Input Buffer is not a TIOVX buffer

ravi-rahul, 2021-08-13:
GST_DEBUG=2 gst-launch-1.0 filesrc location=/opt/test_images/nv12/0000000100.yuv ! video/x-raw, format=NV12, width=1280, height=720, framerate=30/1 ! tiovxcolorconvert ! fakesink sync=true

0:00:00.071380030  1902     0x1b7ba5e0 ERROR              tiovxsiso gsttiovxsiso.c:385:gst_tiovx_siso_transform:<tiovxcolorconvert0> Input Buffer is not a TIOVX buffer
0:00:00.071459870  1902     0x1b7ba5e0 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<filesrc0> error: Internal data stream error.
0:00:00.071499965  1902     0x1b7ba5e0 WARN                 basesrc gstbasesrc.c:3072:gst_base_src_loop:<filesrc0> error: streaming stopped, reason error (-5)

jafet-chaves, 2021-08-20:
We've found that the SIMO and SISO class are not managing the case where the src element upstream does not query a buffer pool, this should be fixable for both classes and we working on a solution for it.

jafet-chaves, 2021-08-20:
The issue for the SISO class has been fixed in #60

ravi-rahul, 2021-08-23:
verified

closed 2021-08-23 by ravi-rahul