TexasInstruments/edgeai-gst-plugins #61: Output is not correct whenever not tiovx buffers are processed by tiovx plugins

ravi-rahul, 2021-08-23:
GST_DEBUG=tiovx*:4 gst-launch-1.0 multifilesrc location=/opt/i420/%010u.yuv index=100 loop=true ! video/x-raw, format=I420, width=1280, height=720, framerate=30/1 ! tiovxcolorconvert !  video/x-raw, format=NV12 ! kmssink
 
 output is not correct for this pipeline
 similar behavior whenever non tiovx buffers are processed

ravi-rahul, 2021-08-23:
the issue can also be reproduced in the application using below steps

1. sync the edge_ai_apps git repo to get the updated video file
2. Modify the config file configs/image_classification.yaml to set the input as video like below
         flows:
             flow0:
                 input: input1 (change input0 to input1)
                 
 3. run the application
      cd apps_python
      ./app_edgeai.py ../configs/image_classification.yaml

ravi-rahul, 2021-08-23:
first run the demo with camera input
after that if we run demo with file input we can see the traces of camera frame in the output

ravi-rahul, 2021-08-24:
Hi @emmadrigal 

I tested the application with fix/memory-copy branch
the issue is not coming, getting correct output

rrcarlosrodriguez, 2021-08-24:
That's great @ravi-rahul ! We will proceed with the merge into develop

Feel free to close this issue whenever you want

closed 2021-08-25 by ravi-rahul