TexasInstruments/edgeai-gst-plugins #75: MEM: ERROR: when multitiple tiovx plugins are used in the same pipeline

ravi-rahul, 2021-09-02:
Pipeline:

GST_DEBUG=tiovx*:4 gst-launch-1.0 videotestsrc is-live=true ! video/x-raw, width=1280, height=720, format=I420 ! tiovxcolorconvert ! video/x-raw, format=NV12 ! tiovxmultiscaler ! video/x-raw, width=454, height=256 ! tiovxcolorconvert ! video/x-raw, format=RGB ! fakesink

Log:

^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:01.359334270
Setting pipeline to NULL ...
0:00:01.437071045  3280     0x11040160 INFO       tiovxcolorconvert gsttiovxcolorconvert.c:613:gst_tiovx_color_convert_deinit_module:<tiovxcolorconvert1> Deinit module
0:00:01.437684100  3280     0x11040160 INFO       tiovxcolorconvert gsttiovxcolorconvert.c:613:gst_tiovx_color_convert_deinit_module:<tiovxcolorconvert0> Deinit module
**MEM: ERROR: Failed to free memory at virt addr = 0xffffa80b0000 !!!**
Freeing pipeline ...
**MEM: ERROR: Failed to free memory at virt addr = 0xffffa82a0000 !!!**

shyam-j, 2021-09-04:
The removed the VX_ZONE_ERRORS but introduces VX_ZONE_WARNING as below. Basically there are some hanging references which are collected during vxReleaseContext. Interestingly I see that this occurs during alternate runs, 
Run1 - no error, no warning,
Run2 - no error, warning present
Run3 - no error, no warning,
Run4 - no error, warning present

Here is an updated pipeline I tried,
gst-launch-1.0 v4l2src device=/dev/video1 io-mode=2 ! image/jpeg, width=1280, height=720 ! jpegdec ! tiovxcolorconvert ! video/x-raw, format=NV12 ! tiovxmultiscaler name=split_01 split_01. ! queue ! video/x-raw, width=454, height=256 ! tiovxcolorconvert out-pool-size=4 ! video/x-raw, format=RGB ! videobox left=115 right=115 top=16 bottom=16 ! tiovxdlpreproc data-type=10 channel-order=0 mean-0=123.675000 mean-1=116.280000 mean-2=103.530000 scale-0=0.017000 scale-1=0.018000 scale-2=0.017000 tensor-format=rgb out-pool-size=4 ! application/x-tensor-tiovx ! fakesink split_01. ! queue ! video/x-raw, width=1280, height=720 ! tiovxcolorconvert target=1 out-pool-size=4 ! video/x-raw, format=RGB ! fakesink

I see warnings as below,
 11036.267508 s:  VX_ZONE_WARNING:[vxReleaseContext:1050] Found a reference 0xffff9b847b30 of type 0000080f at external count 1, internal count 0, releasing it
 11036.267518 s:  VX_ZONE_WARNING:[vxReleaseContext:1054] Unreleased reference name = image_114
 11036.267524 s:  VX_ZONE_WARNING:[vxReleaseContext:1056] Releasing reference now as a part of garbage collection
 11036.267533 s:  VX_ZONE_WARNING:[vxReleaseContext:1050] Found a reference 0xffff9b8b0128 of type 00000813 at external count 1, internal count 0, releasing it
 11036.267540 s:  VX_ZONE_WARNING:[vxReleaseContext:1054] Unreleased reference name = object_array_139
 11036.267545 s:  VX_ZONE_WARNING:[vxReleaseContext:1056] Releasing reference now as a part of garbage collection
 11036.267596 s:  VX_ZONE_WARNING:[vxReleaseContext:1050] Found a reference 0xffff9b8b02c8 of type 00000813 at external count 1, internal count 0, releasing it
 11036.267603 s:  VX_ZONE_WARNING:[vxReleaseContext:1054] Unreleased reference name = object_array_141
 11036.267608 s:  VX_ZONE_WARNING:[vxReleaseContext:1056] Releasing reference now as a part of garbage collection

jafet-chaves, 2021-09-07:
Rahul, please test with the current release (0.2.1) and the following pipeline below. Currently it is needed to specify the `pixel-aspect-ratio=(fraction)1/1` in the caps description in the sink of the tiovxmultiscaler as workaround while we look into the pending issues when running out of memory in this case. For now this pipeline should allow to unblock the testing on your end.

```
GST_DEBUG=*tiovx*:4,GST_TRACER:7 GST_TRACERS=leaks G_DEBUG=fatal-warnings gst-launch-1.0 \
multifilesrc location=/opt/edge_ai_apps/data/images/%04d.jpg loop=true caps="image/jpeg, framerate=1/1" ! jpegdec ! videoscale ! \
"video/x-raw, width=1280, height=720" ! tiovxcolorconvert ! "video/x-raw, format=NV12, pixel-aspect-ratio=(fraction)1/1" ! tiovxmultiscaler ! video/x-raw, width=640, height=480 ! fakesink -e
```

emmadrigal, 2021-09-08:
> Rahul, please test with the current release (0.2.1) and the following pipeline below. Currently it is needed to specify the `pixel-aspect-ratio=(fraction)1/1` in the caps description in the sink pad of the tiovxmultiscaler as a workaround while we look into the pending issues when running out of memory in this specific case. For now this pipeline should allow to unblock the testing on your end.
> 
> ```
> GST_DEBUG=*tiovx*:4,GST_TRACER:7 GST_TRACERS=leaks G_DEBUG=fatal-warnings gst-launch-1.0 \
> multifilesrc location=/opt/edge_ai_apps/data/images/%04d.jpg loop=true caps="image/jpeg, framerate=1/1" ! jpegdec ! videoscale ! \
> "video/x-raw, width=1280, height=720" ! tiovxcolorconvert ! "video/x-raw, format=NV12, pixel-aspect-ratio=(fraction)1/1" ! tiovxmultiscaler ! video/x-raw, width=640, height=480 ! fakesink -e
> ```

This relates to issue #63 not this one

closed 2021-09-09 by ravi-rahul