TexasInstruments/edgeai-gst-plugins #95: with v4l2src and tiovxmultiscaler getting **assertion 'mini_object != NULL' failed**

ravi-rahul, 2021-09-09:
root@j7-evm:/opt/edge_ai_apps/apps_python# gst-launch-1.0 v4l2src device=/dev/video1 ! tiovxcolorconvert ! video/x-raw, format=NV12 ! tiovxmultiscaler ! video/x-raw, width=640, height=480 ! kmssink       
[  505.596955] usb 1-1.3: reset high-speed USB device number 3 using xhci-hcd
APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=5) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
   511.942414 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
   511.942451 s:  VX_ZONE_INIT:Enabled
   511.942457 s:  VX_ZONE_ERROR:Enabled
   511.942463 s:  VX_ZONE_WARNING:Enabled
   511.942950 s:  VX_ZONE_INIT:[tivxInitLocal:111] Initialization Done !!!
   511.943122 s:  VX_ZONE_INIT:[tivxHostInitLocal:86] Initialization Done for HOST !!!
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...

**(gst-launch-1.0:1586): GStreamer-CRITICAL **: 08:26:02.094: gst_mini_object_unref: assertion 'mini_object != NULL' failed
New clock: GstSystemClock**
^Chandling interrupt.

michaelgruner, 2021-09-10:
I confirm this. After some debugging, it boils down [to this line](https://github.com/TexasInstruments/edgeai-gst-plugins/blob/546910a84cb9408b66139afc0e18c37af226cf15/gst-libs/gst/tiovx/gsttiovxsimo.c#L937):

```c
      /* The query response should be the supported caps in the sink
       * from get_caps */
      gst_query_set_caps_result (query, sink_caps);
      gst_caps_unref (sink_caps);
      gst_caps_unref (filter);
      g_list_free_full (src_caps_list, (GDestroyNotify) gst_caps_unref);
```

The filter may be NULL in some occasions, in which case we should not attempt to free them.

dchvs, 2021-09-16:
This assertions is fixed in [PR#99](https://github.com/TexasInstruments/edgeai-gst-plugins/pull/99).

closed 2021-09-16 by dchvs