TexasInstruments/edgeai-gst-plugins #101: tiovxmosaic: MEM error: failed to free memory

aniket-l, 2021-09-21:
Seeing the following error:
MEM: ERROR: Failed to free memory at virt addr = (nil) !!!

when running any pipeline with tiovxmosaic plugin:
gst-launch-1.0 videotestsrc is-live=true num-buffers=5 ! "video/x-raw,format=NV12,width=1920,height=1080" ! \
tiovxmosaic ! filesink location=mosaic_out_nv12_1_input.yuv -e -v

rrcarlosrodriguez, 2021-09-23:
Thanks for the report @aniket-l . @shyam-j This one can be reproduced with the modules test application. The problem happens when using NULL as a background image. You can use the following patch to reproduce the issue with the modules:

```
diff --git a/test/app_tiovx_img_mosaic_module_test.c b/test/app_tiovx_img_mosaic_module_test.c
index a194b35..a3b867f 100644
--- a/test/app_tiovx_img_mosaic_module_test.c
+++ b/test/app_tiovx_img_mosaic_module_test.c
@@ -196,7 +196,7 @@ static vx_status app_create_graph(AppObj *obj)
 
     if((vx_status)VX_SUCCESS == status)
     {
-        status = tiovx_img_mosaic_module_create(obj->graph, &obj->imgMosaicObj, obj->imgMosaicObj.background_image[0], input_arr_user, TIVX_TARGET_VPAC_MSC1);
+        status = tiovx_img_mosaic_module_create(obj->graph, &obj->imgMosaicObj, NULL, input_arr_user, TIVX_TARGET_VPAC_MSC1);
     }
 
     graph_parameter_index = 0;
@@ -210,15 +210,15 @@ static vx_status app_create_graph(AppObj *obj)
         graph_parameter_index++;
     }
 
-    if((vx_status)VX_SUCCESS == status)
-    {   /* Background image index is 2 */
-        status = add_graph_parameter_by_node_index(obj->graph, obj->imgMosaicObj.node, 2);
-        obj->imgMosaicObj.background_graph_parameter_index = graph_parameter_index;
-        graph_parameters_queue_params_list[graph_parameter_index].graph_parameter_index = graph_parameter_index;
-        graph_parameters_queue_params_list[graph_parameter_index].refs_list_size = APP_BUFQ_DEPTH;
-        graph_parameters_queue_params_list[graph_parameter_index].refs_list = (vx_reference*)&obj->imgMosaicObj.background_image[0];
-        graph_parameter_index++;
-    }
+    /* if((vx_status)VX_SUCCESS == status) */
+    /* {   /\* Background image index is 2 *\/ */
+    /*     status = add_graph_parameter_by_node_index(obj->graph, obj->imgMosaicObj.node, 2); */
+    /*     obj->imgMosaicObj.background_graph_parameter_index = graph_parameter_index; */
+    /*     graph_parameters_queue_params_list[graph_parameter_index].graph_parameter_index = graph_parameter_index; */
+    /*     graph_parameters_queue_params_list[graph_parameter_index].refs_list_size = APP_BUFQ_DEPTH; */
+    /*     graph_parameters_queue_params_list[graph_parameter_index].refs_list = (vx_reference*)&obj->imgMosaicObj.background_image[0]; */
+    /*     graph_parameter_index++; */
+    /* } */
 
     if((vx_status)VX_SUCCESS == status)
     {   /* Inputs start from image index 3 */
@@ -275,7 +275,7 @@ static vx_status app_run_graph(AppObj *obj)
     vx_uint32 backSizes[APP_BUFQ_DEPTH][TIOVX_MODULES_MAX_REF_HANDLES];
 
     vxQueryImage(imgMosaicObj->output_image[0], VX_IMAGE_PLANES, &out_num_planes, sizeof(out_num_planes));
-    vxQueryImage(imgMosaicObj->background_image[0], VX_IMAGE_PLANES, &back_num_planes, sizeof(back_num_planes));
+    //vxQueryImage(imgMosaicObj->background_image[0], VX_IMAGE_PLANES, &back_num_planes, sizeof(back_num_planes));
 
     /* These can be moved to app_init() */
     allocate_image_buffers(&imgMosaicObj->inputs[0], inAddr, inSizes);
@@ -283,21 +283,21 @@ static vx_status app_run_graph(AppObj *obj)
     {
         allocate_single_image_buffer(imgMosaicObj->output_image[bufq], outAddr[bufq], outSizes[bufq]);
     }
-    for(bufq = 0; bufq < APP_BUFQ_DEPTH; bufq++)
-    {
-        allocate_single_image_buffer(imgMosaicObj->background_image[bufq], backAddr[bufq], backSizes[bufq]);
-    }
+    /* for(bufq = 0; bufq < APP_BUFQ_DEPTH; bufq++) */
+    /* { */
+    /*     allocate_single_image_buffer(imgMosaicObj->background_image[bufq], backAddr[bufq], backSizes[bufq]); */
+    /* } */
 
     bufq = 0;
     assign_image_buffers(&imgMosaicObj->inputs[0], inAddr[bufq], inSizes[bufq], bufq);
     assign_single_image_buffer(imgMosaicObj->output_image[0], outAddr[bufq], outSizes[bufq], out_num_planes);
-    assign_single_image_buffer(imgMosaicObj->background_image[0], outAddr[bufq], outSizes[bufq], back_num_planes);
+    /* assign_single_image_buffer(imgMosaicObj->background_image[0], outAddr[bufq], outSizes[bufq], back_num_planes); */
 
     //status = vxProcessGraph(obj->graph);
 
     release_image_buffers(&imgMosaicObj->inputs[0], inAddr[bufq], inSizes[bufq], bufq);
     release_single_image_buffer(imgMosaicObj->output_image[0], outAddr[bufq], outSizes[bufq], out_num_planes);
-    release_single_image_buffer(imgMosaicObj->background_image[0], outAddr[bufq], outSizes[bufq], back_num_planes);
+    /* release_single_image_buffer(imgMosaicObj->background_image[0], outAddr[bufq], outSizes[bufq], back_num_planes); */
 
     /* These can move to deinit() */
     delete_image_buffers(&imgMosaicObj->inputs[0], inAddr, inSizes);
@@ -305,10 +305,10 @@ static vx_status app_run_graph(AppObj *obj)
     {
         delete_single_image_buffer(imgMosaicObj->output_image[bufq], outAddr[bufq], outSizes[bufq]);
     }
-    for(bufq = 0; bufq < APP_BUFQ_DEPTH; bufq++)
-    {
-        delete_single_image_buffer(imgMosaicObj->background_image[bufq], backAddr[bufq], backSizes[bufq]);
-    }
+    /* for(bufq = 0; bufq < APP_BUFQ_DEPTH; bufq++) */
+    /* { */
+    /*     delete_single_image_buffer(imgMosaicObj->background_image[bufq], backAddr[bufq], backSizes[bufq]); */
+    /* } */
 
     return status;
 }

```

shyam-j, 2021-09-24:
This issue is fixed now and merged to main branch. Please take the latest from main.
https://github.com/TexasInstruments/edgeai-tiovx-modules/commit/5c7d8c9eab7cd483d9a94eb9c3da5244a25f394d

closed 2021-09-24 by shyam-j