TexasInstruments/edgeai-gst-plugins #107: tiovxmosaic: Unable to launch multiple instances of tiovxmosaic

aniket-l, 2021-09-23:
When using multiple instances of tiovxmosaic
> gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12, width=1280, height=720 ! tiovxmosaic name=mosaic_0 ! video/x-raw,format=NV12, width=1920, height=1080 ! kmssink sync=false driver-name=tidss videotestsrc ! video/x-raw,format=NV12, width=1280, height=720 ! tiovxmosaic name=mosaic_1 ! video/x-raw,format=NV12, width=1920, height=1080 ! jpegenc ! qtmux ! filesink location=output_video.mov

Error logs:
> Pipeline is PREROLLING ...
> MEM: ERROR: Failed to free memory at virt addr = (nil) !!!
>    853.194866 s:  VX_ZONE_ERROR:[vxReleaseKernel:43] Invalid kernel reference
>    853.194896 s:  VX_ZONE_ERROR:[vxGetStatus:713] Reference is NULL
> [IMG-MOSAIC-MODULE] Unable to create kernel with 1 inputs!
>    853.194955 s:  VX_ZONE_ERROR:[vxCreateNodeByStructure:96] failed to retrieve kernel enum 0
>    853.194976 s:  VX_ZONE_ERROR:[vxGetStatus:713] Reference is NULL
> [IMG-MOSAIC-MODULE] Unable to create mosaic node! 
> ERROR: from element /GstPipeline:pipeline0/GstTIOVXMosaic:mosaic_1: Unable to init TIOVX module
> Additional debug info:
> ../gst-libs/gst/tiovx/gsttiovxmiso.c(1096): gst_tiovx_miso_negotiated_src_caps (): /GstPipeline:pipeline0/GstTIOVXMosaic:mosaic_1
> ERROR: pipeline doesn't want to preroll.

rrcarlosrodriguez, 2021-09-23:
Hi @aniket-l , thanks for the report, I was able to reproduce the issue!

@shyam-j I was also able to reproduce this issue with the modules in a standalone application. It looks like just by using the mosaic module twice, the issue appears. For example, I applied the following patch at the top of the modules repo:

```
diff --git a/test/app_tiovx_img_mosaic_module_test.c b/test/app_tiovx_img_mosaic_module_test.c
index a194b35..11859a8 100644
--- a/test/app_tiovx_img_mosaic_module_test.c
+++ b/test/app_tiovx_img_mosaic_module_test.c
@@ -82,6 +82,7 @@ typedef struct {
 } AppObj;
 
 AppObj gAppObj;
+AppObj gAppObj1;
 
 static vx_status app_init(AppObj *obj);
 static void app_deinit(AppObj *obj);
@@ -93,33 +94,60 @@ static void app_delete_graph(AppObj *obj);
 vx_status app_modules_img_mosaic_test(vx_int32 argc, vx_char* argv[])
 {
     AppObj *obj = &gAppObj;
+    AppObj *obj1 = &gAppObj1;
     vx_status status = VX_SUCCESS;
 
     status = app_init(obj);
     APP_PRINTF("App Init Done! \n");
 
+    if(status == VX_SUCCESS)
+    {
+        status = app_init(obj1);
+        APP_PRINTF("Second App Init Done! \n");
+    }
     if(status == VX_SUCCESS)
     {
         status = app_create_graph(obj);
         APP_PRINTF("App Create Graph Done! \n");
     }
     if(status == VX_SUCCESS)
+    {
+        status = app_create_graph(obj1);
+        APP_PRINTF("Second App Create Graph Done! \n");
+    }
+    if(status == VX_SUCCESS)
     {
         status = app_verify_graph(obj);
         APP_PRINTF("App Verify Graph Done! \n");
     }
+    if(status == VX_SUCCESS)
+    {
+        status = app_verify_graph(obj1);
+        APP_PRINTF("Second App Verify Graph Done! \n");
+    }
     if (status == VX_SUCCESS)
     {
         status = app_run_graph(obj);
         APP_PRINTF("App Run Graph Done! \n");
     }
+    if (status == VX_SUCCESS)
+    {
+        status = app_run_graph(obj1);
+        APP_PRINTF("Second App Run Graph Done! \n");
+    }
 
     app_delete_graph(obj);
     APP_PRINTF("App Delete Graph Done! \n");
 
+    app_delete_graph(obj1);
+    APP_PRINTF("Second App Delete Graph Done! \n");
+
     app_deinit(obj);
     APP_PRINTF("App De-init Done! \n");
 
+    app_deinit(obj1);
+    APP_PRINTF("Second App De-init Done! \n");
+
     return status;
 }

```

And I got exactly the same issues obtained with the pipeline:

```
APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=4) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
108367.313956 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
108367.314039 s:  VX_ZONE_INIT:Enabled
108367.314050 s:  VX_ZONE_ERROR:Enabled
108367.314058 s:  VX_ZONE_WARNING:Enabled
108367.314501 s:  VX_ZONE_INIT:[tivxInitLocal:111] Initialization Done !!!
108367.314729 s:  VX_ZONE_INIT:[tivxHostInitLocal:86] Initialization Done for HOST !!!
Running image mosaic module test
108367.315863 s:  VX_ZONE_ERROR:[vxReleaseKernel:43] Invalid kernel reference
108367.315880 s:  VX_ZONE_ERROR:[vxGetStatus:713] Reference is NULL
[IMG-MOSAIC-MODULE] Unable to create kernel with 1 inputs!
108367.315933 s:  VX_ZONE_ERROR:[ownReleaseReferenceInt:307] Invalid reference
108367.315951 s:  VX_ZONE_ERROR:[ownReleaseReferenceInt:307] Invalid reference
All tests complete!
108367.316267 s:  VX_ZONE_INIT:[tivxHostDeInitLocal:100] De-Initialization Done for HOST !!!
108367.320615 s:  VX_ZONE_INIT:[tivxDeInitLocal:155] De-Initialization Done !!!
APP: Deinit ... !!!
REMOTE_SERVICE: Deinit ... !!!
REMOTE_SERVICE: Deinit ... Done !!!
IPC: Deinit ... !!!
IPC: DeInit ... Done !!!
MEM: Deinit ... !!!
MEM: Alloc's: 2 alloc's of 2848 bytes 
MEM: Free's : 2 free's  of 2848 bytes 
MEM: Open's : 0 allocs  of 0 bytes 
MEM: Deinit ... Done !!!
APP: Deinit ... Done !!!
```

@shyam-j could you please double-check this behavior on your end?

shyam-j, 2021-09-24:
Thanks @rrcarlosrodriguez @aniket-l , this bug requires some change in TIOVX. I have found the issue but checking with experts on the fix. Will keep you posted.

aniket-l, 2021-12-13:
Fixed in 8.1 SDK

closed 2021-12-13 by aniket-l