TexasInstruments/edgeai-gst-plugins #110: tiovxmosaic: to support max frame rate

aniket-l, 2021-09-24:
> videotestsrc pattern=ball is-live=true ! video/x-raw,format=NV12, width=640, height=360, framerate=30/1 ! queue ! mosaic.sink_0 \
videotestsrc pattern=ball is-live=true ! video/x-raw,format=NV12, width=640, height=360, framerate=1/1 ! queue ! mosaic.sink_1 \
tiovxmosaic name=mosaic \
sink_0::startx=0   sink_0::starty=0   sink_0::width=640  sink_0::height=360 \
sink_1::startx=640 sink_1::starty=0   sink_1::width=640  sink_1::height=360 \
! video/x-raw,format=NV12, width=1920, height=1080 ! kmssink sync=false driver-name=tidss

The above pipeline runs at 1 FPS. Want it to run at 30 FPS (or whatever the max frame rate is for the input streams)

m-herrera, 2021-11-22:
@aniket-l Please verify if the issue is solved by #146; this fix is already merged into develop.

aniket-l, 2021-11-26:
The develop branch has not rolled back the 2A integration changes so I could not test using that branch. I cherry-picked the 2 commits into current main branch and tested the sample pipeline above. 
It did not seem to work correctly: I see the faster channel running at the desired frame rate but the slower one is completely frozen. This is happening for some combinations of the framerates (for example the one in the above pipeline). Are you not getting this issue?

m-herrera, 2021-11-26:
You are right, I am able to reproduce the issue. However, I can only reproduce it while using kmssink and with some framerates (as you mentioned), I will take a closer look at it to see what is going on. 

Thanks for the observation.

m-herrera, 2021-12-01:
@aniket-l Please test our latest fix in branch `fix/miso-max-framerate-error` to see if you get the desired output on your end. Once you test it we can merge PR #151 with those changes.