TexasInstruments/edgeai-gst-plugins #115: TIOVXDLPreProc: High input resolutions fails allocating TIVX memory

dchvs, 2021-09-28:
```bash
gst-launch-1.0  videotestsrc ! video/x-raw,width=7680,height=4320 ! tiovxdlpreproc ! application/x-tensor-tiovx ! fakesink
APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=4) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
 33764.526450 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
 33764.526533 s:  VX_ZONE_INIT:Enabled
 33764.526545 s:  VX_ZONE_ERROR:Enabled
 33764.526554 s:  VX_ZONE_WARNING:Enabled
 33764.526978 s:  VX_ZONE_INIT:[tivxInitLocal:111] Initialization Done !!!
 33764.527190 s:  VX_ZONE_INIT:[tivxHostInit:48] Initialization Done for HOST !!!
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.534308 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
 33764.534342 s:  VX_ZONE_ERROR:[ownAllocTensorBuffer:83] Could not allocate tensor memory
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.534378 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
 33764.534395 s:  VX_ZONE_ERROR:[ownAllocTensorBuffer:83] Could not allocate tensor memory
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.534426 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
 33764.534472 s:  VX_ZONE_ERROR:[ownAllocTensorBuffer:83] Could not allocate tensor memory
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.534517 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.534917 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
 33764.534929 s:  VX_ZONE_ERROR:[ownAllocImageBuffer:309] could not allocate memory
 33764.534938 s:  VX_ZONE_ERROR:[ownCopyAndMapCheckParams:672] image allocation failed
MEM: ERROR: Alloc failed with status = 12 !!!
 33764.535063 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
ERROR: from element /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0: Internal data stream error.
Additional debug info:
../gstreamer-1.16.3/libs/gst/base/gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
 33764.536528 s:  VX_ZONE_WARNING:[vxReleaseContext:1012] Found a reference 0xffffa9b2a998 of type 00000813 at external count 1, internal count 0, releasing it
 33764.536546 s:  VX_ZONE_WARNING:[vxReleaseContext:1016] Unreleased reference name = object_array_104
 33764.536570 s:  VX_ZONE_WARNING:[vxReleaseContext:1018] Releasing reference now as a part of garbage collection
 33764.536825 s:  VX_ZONE_WARNING:[vxReleaseContext:1012] Found a reference 0xffffa9b2ab38 of type 00000813 at external count 1, internal count 0, releasing it
 33764.536836 s:  VX_ZONE_WARNING:[vxReleaseContext:1016] Unreleased reference name = object_array_106
 33764.536843 s:  VX_ZONE_WARNING:[vxReleaseContext:1018] Releasing reference now as a part of garbage collection
 33764.536951 s:  VX_ZONE_INIT:[tivxHostDeInit:56] De-Initialization Done for HOST !!!
 33764.541269 s:  VX_ZONE_INIT:[tivxDeInitLocal:155] De-Initialization Done !!!
APP: Deinit ... !!!
REMOTE_SERVICE: Deinit ... !!!
REMOTE_SERVICE: Deinit ... Done !!!
IPC: Deinit ... !!!
IPC: DeInit ... Done !!!
MEM: Deinit ... !!!
MEM: Alloc's: 5 alloc's of 995328052 bytes 
MEM: Free's : 5 free's  of 995328052 bytes 
MEM: Open's : 0 allocs  of 0 bytes 
MEM: Deinit ... Done !!!
APP: Deinit ... Done !!!
```

m-herrera, 2021-12-02:
This issue can be reproduced in the modules by setting the test application to use the same parameters and duplicating the number of allocs to simulate a pool size of 2, the following patch shows those changes.

```
---
 test/app_tiovx_dl_pre_proc_module_test.c | 8 +++++---
 1 file changed, 5 insertions(+), 3 deletions(-)

diff --git a/test/app_tiovx_dl_pre_proc_module_test.c b/test/app_tiovx_dl_pre_proc_module_test.c
index 1e751e6..c366bbe 100644
--- a/test/app_tiovx_dl_pre_proc_module_test.c
+++ b/test/app_tiovx_dl_pre_proc_module_test.c
@@ -66,8 +66,8 @@
 #define APP_BUFQ_DEPTH   (1)
 #define APP_NUM_CH       (1)
 
-#define IMAGE_WIDTH      (640)
-#define IMAGE_HEIGHT     (480)
+#define IMAGE_WIDTH      (7680)
+#define IMAGE_HEIGHT     (4320)
 
 typedef struct {
 
@@ -163,7 +163,7 @@ static vx_status app_init(AppObj *obj)
         dlPreProcObj->input.height = IMAGE_HEIGHT;
 
         dlPreProcObj->output.bufq_depth = APP_BUFQ_DEPTH;
-        dlPreProcObj->output.datatype = VX_TYPE_UINT8;
+        dlPreProcObj->output.datatype = VX_TYPE_FLOAT32;
         dlPreProcObj->output.num_dims = 3;
         /* This can be updated based on NCHW or NHWC */
         dlPreProcObj->output.dim_sizes[0] = IMAGE_WIDTH;
@@ -282,6 +282,8 @@ static vx_status app_run_graph(AppObj *obj)
 
     /* These can be moved to app_init() */
     allocate_image_buffers(&dlPreProcObj->input, inAddr, inSizes);
+    allocate_image_buffers(&dlPreProcObj->input, inAddr, inSizes);
+    allocate_tensor_buffers(&dlPreProcObj->output, outAddr, outSizes);
     allocate_tensor_buffers(&dlPreProcObj->output, outAddr, outSizes);
 
     bufq = 0;
```

After running the test app it show the same error:

```
MEM: ERROR: Alloc failed with status = 12 !!!
```

ksmu, 2025-07-07:
Hi @dchvs99 , 
I am facing with a similar error running my model transformed for c7x on am62a. I cannot find the final verdict here. How did you proceed?