TexasInstruments/edgeai-gst-plugins #139: Backport GStreamer elements to work with Ubuntu 18.04 and GStreamer 1.14.5

rrcarlosrodriguez, 2021-11-03:
TI will provide a docker image with Ubuntu 18.04 for RidgeRun to use to verify elements functionality

m-herrera, 2021-11-23:
@shyam-j we ran into into some issues while executing the tiovx modules test app inside the Ubuntu 18.04 container.

It appears to be an issue with the version of glibc, Ubuntu 18.04 comes with version  2.27 and I get an error requesting at least 2.29.

The issue seems to be originated from `libtivision_apps.so`, if I try to link the library alone as a test, I get the follwoing error.
```
[docker] root@j7-2:/opt/edge_ai_apps# ld -ltivision_apps
ld: warning: cannot find entry symbol _start; not setting start address
//usr/lib/libtivision_apps.so: undefined reference to `pow@GLIBC_2.29'
```

I updated the version of glib inside the container (although it is not recommended and I am sure it broke some packages). And managed to get it to work with the modules as well as the gstreamer elements (with some minor changes).

Is updating glib the expected approach, or do you have an alternative in order to run the modules with Ubuntu 18.04?

If glib 2.29 is a must, then maybe the docker image can provide it.

shyam-j, 2021-11-24:
@m-herrera We certainly do not want to change glib version but we can always build Ubuntu 18.04 for aarch64 using a QEMU setup. I will connect you with JuneChul from TI who has done this.