TexasInstruments/edgeai-gst-plugins #166: Random hang observed while running edge_ai_apps with edgeai-gst-plugins

shyam-j, 2021-12-23:
This happens with 8.1 SDK RC when we run edge_ai_apps which uses edgeai-tiovx-modules

Please follow below steps to reproduce. Due to the nature of the issue, the hang can occur either the very first time after boot or might need several reboots to reproduce the issue. However if the demo runs fine and exits fine after first launch, then the demo will continue to run without any problems. 

Also when we exit the hang by pressing Ctrl+<backslash> we can run the same application again and it will continue to run without any issues. 

1. Install the 8.1 SDK (RC or public release) on SD card
2. Run the setup_script.sh from /opt/edge_ai_apps folder only for the first time
3. Run the out of box python demo /opt/edge_ai_apps/apps_python# ./app_edgeai ../configs/image_classification -n
4. At this point the demo might come up but hang right after by printing messages like these,

[ 3036.951557] virtio_rpmsg_bus virtio3: creating channel rpmsg_chrdev addr 0xd
[ 3036.958709] virtio_rpmsg_bus virtio3: channel rpmsg_chrdev:ffffffff:d already exist
[ 3036.966479] virtio_rpmsg_bus virtio3: rpmsg_create_channel failed

5. Once the demo hangs, we can recover by killing the process or pressing Ctrl+<backslash>
6. We can run the same demo again, without restarting the board and this time the demo and other demos should run without any issues.

m-herrera, 2021-12-28:
@shyam-j I am not able to reproduce this error. I rebooted and tested multiple times in both the plugins main and develop branches. The application is not hanging; however, I get two errors, one regarding the mosaic caps, and a memory error that results in a segmentation fault after a long time.
I will provide the logs for these errors below; however, we should address those separately.
```
root@j7-sk2:/opt/edge_ai_apps/apps_python# GST_DEBUG=2 ./app_edgeai.py ../configs/image_classification.yaml -n
2021-12-28 17:11:45,674 INFO Could not find libdlr.so in model artifact. Using dlr from /usr/dlr/libdlr.so
Calling appInit() in TIDL-RT!
APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=4) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
586398.214937 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
586398.214979 s:  VX_ZONE_INIT:Enabled
586398.214987 s:  VX_ZONE_ERROR:Enabled
586398.214999 s:  VX_ZONE_WARNING:Enabled
586398.215474 s:  VX_ZONE_INIT:[tivxInitLocal:111] Initialization Done !!!
586398.215678 s:  VX_ZONE_INIT:[tivxHostInitLocal:86] Initialization Done for HOST !!!
[GST SRC STR]
[FLOW 0]
multifilesrc location=/opt/edge_ai_apps/data/images/%04d.jpg loop=true caps=image/jpeg,framerate=1/1  ! jpegdec  ! videoscale ! video/x-raw, width=1280, height=720 ! tiovxcolorconvert ! video/x-raw, format=NV12 ! tiovxmultiscaler name=split_01 
split_01. ! queue ! video/x-raw, width=454, height=256 ! tiovxcolorconvert out-pool-size=4 ! video/x-raw, format=RGB ! videobox left=115 right=115 top=16 bottom=16 ! tiovxdlpreproc data-type=10 channel-order=0 mean-0=123.675000 mean-1=116.280000 mean-2=103.530000 scale-0=0.017000 scale-1=0.018000 scale-2=0.017000 tensor-format=rgb out-pool-size=4 ! application/x-tensor-tiovx ! appsink name=pre_0 max-buffers=2 drop=true 
split_01. ! queue ! video/x-raw, width=1280, height=720 ! tiovxcolorconvert target=1 out-pool-size=4 ! video/x-raw, format=RGB ! appsink name=sen_0 max-buffers=2 drop=true 

[GST SINK STR]
appsrc format=GST_FORMAT_TIME is-live=true block=true do-timestamp=true name=post_0 ! tiovxcolorconvert ! video/x-raw,format=NV12, width=1280, height=720 ! queue ! mosaic_0.sink_0 
appsrc format=GST_FORMAT_TIME block=true num-buffers=1 name=background_0 ! tiovxcolorconvert ! video/x-raw,format=NV12, width=1920, height=1080 ! queue ! mosaic_0.background 
tiovxmosaic name=mosaic_0
sink_0::startx=320  sink_0::starty=180  sink_0::width=1280   sink_0::height=720  
! video/x-raw,format=NV12, width=1920, height=1080 ! kmssink sync=false driver-name=tidss 

0:00:00.951055057  3121     0x40b56370 WARN            videodecoder gstvideodecoder.c:2425:gst_video_decoder_chain:<jpegdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
0:00:01.061326315  3121     0x40b56000 ERROR            tiovxmosaic gsttiovxmosaic.c:973:gst_tiovx_mosaic_fixate_caps:<mosaic_0> Width and height couldn't be fixated to : video/x-raw, format=(string)NV12, width=(int)1920, height=(int)1080, framerate=(fraction)25/1;
0:00:01.116863331  3121     0x40b56000 WARN            kmsallocator gstkmsallocator.c:549:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
0:00:02.107805920  3121     0x40b56000 WARN            kmsallocator gstkmsallocator.c:549:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
 MEM: ERROR: Alloc failed with status = 12 !!!
586496.494908 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
586496.494940 s:  VX_ZONE_ERROR:[ownAllocImageBuffer:309] could not allocate memory
586496.494960 s:  VX_ZONE_ERROR:[ownCopyAndMapCheckParams:672] image allocation failed
586496.494979 s:  VX_ZONE_ERROR:[vxUnmapImagePatch:2059] map ID is greater than the maximum image maps
MEM: ERROR: Alloc failed with status = 12 !!!
586496.495015 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
586496.495032 s:  VX_ZONE_ERROR:[ownAllocImageBuffer:309] could not allocate memory
586496.495074 s:  VX_ZONE_ERROR:[ownCopyAndMapCheckParams:672] image allocation failed
586496.495092 s:  VX_ZONE_ERROR:[vxUnmapImagePatch:2059] map ID is greater than the maximum image maps
MEM: ERROR: Alloc failed with status = 12 !!!
586496.495183 s:  VX_ZONE_ERROR:[tivxMemBufferAlloc:80] Shared mem ptr allocation failed
0:01:39.046306621  3121     0x40b56370 ERROR         tiovxallocator gsttiovxallocator.c:176:gst_tiovx_allocator_alloc:<tiovxallocator509> Unable to allocate dma memory buffer: -8
0:01:39.046331992  3121     0x40b56370 ERROR        tiovxbufferpool gsttiovxbufferpool.c:240:gst_tiovx_buffer_pool_alloc_buffer:<tiovximagebufferpool407> Unable to allocate memory
0:01:39.046357187  3121     0x40b56370 WARN              bufferpool gstbufferpool.c:305:do_alloc_buffer:<tiovximagebufferpool407> alloc function failed
0:01:39.046380172  3121     0x40b56370 WARN              bufferpool gstbufferpool.c:338:default_start:<tiovximagebufferpool407> failed to allocate buffer
0:01:39.046413977  3121     0x40b56370 ERROR             bufferpool gstbufferpool.c:559:gst_buffer_pool_set_active:<tiovximagebufferpool407> start failed
0:01:39.046443633  3121     0x40b56370 ERROR               tiovxpad gsttiovxpad.c:418:gst_tiovx_pad_acquire_buffer:<split_01:src_1> Unable to acquire buffer from pool: -2
0:01:39.046473168  3121     0x40b56370 ERROR              tiovxsimo gsttiovxsimo.c:1094:gst_tiovx_simo_pads_to_vx_references:<split_01> Unable to acquire buffer from pad: 0x408ecbf0
Segmentation fault (core dumped)
```

Regarding the hang issue; is this behavior seen in the main branch of the demos, the modules and the plugins? Maybe I am missing a patch or testing in the wrong branch

shyam-j, 2022-01-05:
@m-herrera This issue is easily reproducible by running just the multi-scaler plugin with a camera input as below on the latest 8.1 SDK that I have shared via CDDS.

This happens randomly. It might just run without any issues so you will have to reboot couple of times and rerun 1till you see the issue.

gst-launch-1.0 v4l2src device=/dev/video2 io-mode=2 ! image/jpeg, width=1280, height=720 ! jpegdec ! videoconvert ! video/x-raw, format=NV12 ! tiovxmultiscaler ! video/x-raw, width=454, height=256 ! fakesink

rrcarlosrodriguez, 2022-01-05:
This issue happens only after boot, and sometimes, not everytime.

@m-herrera  There is an executable that you can use to verify all the services are up. @shyam-j please provide the command.

aniket-l, 2022-01-05:
I also observed the hang using videotestsrc:

gst-launch-1.0 videotestsrc is-live=true ! video/x-raw, format=NV12, width=1280, height=720 ! tiovxmultiscaler ! video/x-raw, width=454, height=256 ! kmssink

m-herrera, 2022-01-05:
@shyam-j I am able to reproduce the error where the pipeline hangs and I have to kill the process; however, I am not getting these messages you reported:
 ```
[ 3036.951557] virtio_rpmsg_bus virtio3: creating channel rpmsg_chrdev addr 0xd
[ 3036.958709] virtio_rpmsg_bus virtio3: channel rpmsg_chrdev:ffffffff:d already exist
[ 3036.966479] virtio_rpmsg_bus virtio3: rpmsg_create_channel failed
 ```

Do these only come up from the app or should I be getting them after closing the pipeline as well?

m-herrera, 2022-01-05:
Below are a few of our findings and tests.

1. The issue is present on 8.1 as well as 8.0.1 (our version with patches, haven't tested on a clean 8.0 installation).
2. We tested multiple (more common) resolutions and the issue is still there.
3. The issue was not introduced in the latest release. We am getting the same behavior before the multi-channel support was added. 
4. Most of the time the hang only occurs on the first pipeline after boot; however, We got the pipeline to hang on the second, and even third pipeline after killing previous pipelines (this is not very common).
5. The issue is not only for the multiscaler; we tested the other elements and got the error for: `tiovxldc`, `tiovxmosaic`, `tiovxmultiscaler`. All of which use VPAC targets (might not be related, but I think it is worth noting).

These are the basic pipelines I used:
- tiovxldc
```
GST_DEBUG=*ti*:6 gst-launch-1.0 -e videotestsrc !  video/x-raw, format=NV12, width=1936, height=1096 ! tiovxldc dcc-file="/opt/imaging/imx390/dcc_ldc_wdr.bin" sensor-name="SENSOR_SONY_IMX390_UB953_D3"  ! video/x-raw,width=1920,height=1080 ! fakesink
```

- tiovxmultiscaler
```
GST_DEBUG=*ti*:LOG gst-launch-1.0 videotestsrc is-live=true ! video/x-raw, format=NV12, width=1280, height=720 ! tiovxmultiscaler ! video/x-raw, width=1280, height=720 ! fakesink
```

- tiovxmosaic
```
GST_DEBUG=*ti*:LOG gst-launch-1.0 videotestsrc is-live=true ! "video/x-raw,format=NV12,width=1920,height=1080" ! tiovxmosaic ! fakesink
```

@shyam-j  If you could provide the executable for seeing the services and other kernel logs, that would be great for our next debugging steps.

m-herrera, 2022-01-05:
This issue is reproducible in early plugin releases (v0.2.0)

m-herrera, 2022-01-06:
Removing the mutex locks from the context singleton does not solve the issue.

Also, the pipeline does process a few times before hanging. The log indicates that the pipeline is hanging on one of the vxWaitGraph calls, waiting for the scheduled graph to complete.

ravi-rahul, 2023-03-01:
This issue is not reproduced in latest SDKs (8.4, 8.5, 8.6 ....)

closed 2023-03-01 by ravi-rahul