TexasInstruments/edgeai-gst-plugins #168: Artifacts on background when using pad

m-herrera, 2021-12-28:
The following pipeline produces artifacts on the background every other frame on the develop branch:

```
gst-launch-1.0 \
videotestsrc is-live=true num-buffers=5 ! "video/x-raw,format=NV12, width=640, height=480,framerate=30/1" ! queue ! mosaic.sink_0 \
videotestsrc is-live=true pattern=ball num-buffers=5 ! "video/x-raw,format=NV12, width=640, height=480, framerate=30/1" ! queue ! mosaic.sink_1 \
videotestsrc pattern=ball num-buffers=2 ! "video/x-raw,format=NV12, width=1290, height=970, framerate=30/1" ! queue ! mosaic.background \
tiovxmosaic name=mosaic \
sink_0::startx=650 \
sink_0::starty=0 \
sink_1::startx=0 \
sink_1::starty=490 \
! "video/x-raw,format=NV12" ! videoconvert ! pngenc ! multifilesink location=/tmp/image_%d.png
```

This only happens if the source of the background pad has `num-buffers=2`. However, when using more buffers the background stream only alternates between two frames.

m-herrera, 2021-12-28:
This issue is not seen in the main branch; at least not with the pipeline above. And it is a lot easier to reproduce on the `fix/miso-max-framerate-error` branch.

aniket-l, 2022-02-09:
Any updates on this issue?

m-herrera, 2022-02-15:
Hi @aniket-l

There are no updates regarding this issue, since we have been working on other issues with higher priority; please let us know if this is of high priority to you so we can plan accordingly