TexasInstruments/edgeai-gst-plugins #173: tiovxmultiscalar padding issue

aniket-l, 2022-01-05:
Working pipelines:
`GST_DEBUG=2 gst-launch-1.0 filesrc location=/opt/edge_ai_apps/data/videos/video_0000_h265.mp4  ! qtdemux  ! h265parse ! v4l2h265dec ! video/x-raw, format=NV12 ! tiovxmultiscaler ! video/x-raw, width=640, height=480 ! kmssink`
`GST_DEBUG=2 gst-launch-1.0 filesrc location=/opt/edge_ai_apps/data/videos/video_0000_h265.mp4  ! qtdemux  ! h265parse ! v4l2h265dec ! video/x-raw, format=NV12 ! tee ! videoscale ! video/x-raw, width=640, height=480 ! kmssink
`

This gives following error:
`GST_DEBUG=2 gst-launch-1.0 filesrc location=/opt/edge_ai_apps/data/videos/video_0000_h265.mp4  ! qtdemux  ! h265parse ! v4l2h265dec ! video/x-raw, format=NV12 ! tee ! tiovxmultiscaler ! video/x-raw, width=640, height=480 ! kmssink`

> Setting pipeline to PAUSED ...
> 0:00:01.020965520  1722     0x23094b90 WARN                 basesrc gstbasesrc.c:3600:gst_base_src_start_complete:<filesrc0> pad not activated yet
> Pipeline is PREROLLING ...
> 0:00:01.021716215  1722     0x23098e80 WARN                 qtdemux qtdemux.c:3250:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
> 0:00:01.029515775  1722     0x23098e80 WARN                    v4l2 gstv4l2object.c:4214:gst_v4l2_object_probe_caps:<v4l2h265dec0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Invalid argument
> 0:00:01.035010670  1722     0x23098e80 WARN            v4l2videodec gstv4l2videodec.c:813:gst_v4l2_video_dec_decide_allocation:<v4l2h265dec0> Duration invalid, not setting latency
> 0:00:01.038376465  1722     0x23098e80 WARN          v4l2bufferpool gstv4l2bufferpool.c:809:gst_v4l2_buffer_pool_start:<v4l2h265dec0:pool:src> Uncertain or not enough buffers, enabling copy threshold
> 0:00:01.064579150  1722 0xffffa4029400 ERROR               tiovxpad gsttiovxbufferutils.c:239:gst_tiovx_buffer_copy: Copy and input size don't match. Copy size is :1382400 and input size is : 1474560d
> 0:00:01.064713025  1722 0xffffa4029400 ERROR               tiovxpad gsttiovxbufferutils.c:396:gst_tiovx_validate_tiovx_buffer: Failure when copying input buffer from pool
> 0:00:01.064765680  1722 0xffffa4029400 ERROR               tiovxpad gsttiovxpad.c:398:gst_tiovx_pad_chain:<tiovxmultiscaler0:sink> Unable to validate buffer
> 0:00:01.064791400  1722 0xffffa4029400 ERROR              tiovxsimo gsttiovxsimo.c:1161:gst_tiovx_simo_chain:<tiovxmultiscaler0:sink> Pad's chain function failed
> 0:00:01.065527215  1722     0x23098e80 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: Internal data stream error.
> 0:00:01.065643580  1722     0x23098e80 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: streaming stopped, reason error (-5)
> ERROR: from element /GstPipeline:pipeline0/GstQTDemux:qtdemux0: Internal data stream error.
> Additional debug info:
> ../gst-plugins-good-1.16.3/gst/isomp4/qtdemux.c(6619): gst_qtdemux_loop (): /GstPipeline:pipeline0/GstQTDemux:qtdemux0:
> streaming stopped, reason error (-5)
> ERROR: pipeline doesn't want to preroll.
> Setting pipeline to NULL ...
> Freeing pipeline ...

An observation about the error:
1382400 = 1280*720*1.5 (NV12) is expected
1474560 = 1280*768*1.5  :  768 is 64byte aligned

m-herrera, 2022-01-10:
@aniket-l Please test the fix from branch [fix/memory-size-mismatch](https://github.com/TexasInstruments/edgeai-gst-plugins/tree/fix/memory-size-mismatch).

aniket-l, 2022-01-12:
Verified.

closed 2022-01-12 by aniket-l