TexasInstruments/edgeai-gst-plugins #174: tiovxmosaic EOS buffer PTS handling

aniket-l, 2022-01-07:
On receiving EOS, mosaic is passing a buffer with PTS set to Gst.CLOCK_TIME_NONE. This is causing an error from qtmux when trying to write to a video file.

Reproduce:
`GST_DEBUG=2 gst-launch-1.0 videotestsrc num-buffers=30 do-timestamp=true ! video/x-raw, format=NV12, width=1280, height=720 ! queue ! identity check-imperfect-timestamp=true name=sniffer1 ! mosaic_0.sink_0  videotestsrc num-buffers=1 do-timestamp=true ! video/x-raw,format=NV12, width=1920, height=1080 ! queue ! mosaic_0.background   tiovxmosaic name=mosaic_0  sink_0::startx=320  sink_0::starty=180  sink_0::width=1280   sink_0::height=720  ! video/x-raw,format=NV12, width=1920, height=1080 ! identity check-imperfect-timestamp=true name=sniffer2 ! jpegenc ! qtmux !  filesink location=a.mov`

Error log :
> 0:00:00.959759120  6248     0x40606f20 WARN                   qtmux gstqtmux.c:4832:gst_qt_mux_add_buffer:<qtmux0> error: Buffer has no PTS.
> ERROR: from element /GstPipeline:pipeline0/GstQTMux:qtmux0: Could not multiplex stream.
> Additional debug info:
> ../gst-plugins-good-1.16.3/gst/isomp4/gstqtmux.c(4832): gst_qt_mux_add_buffer (): /GstPipeline:pipeline0/GstQTMux:qtmux0:
> Buffer has no PTS.

The result is that the video file is corrupt/not-playable.

m-herrera, 2022-01-07:
@aniket-l this issue was caused because the miso base class was pushing a duplicate frame on eos, and said frame didn't have a valid PTS as you mentioned.

Please verify the fix in branch [fix/duplicate-frame-on-eos](https://github.com/TexasInstruments/edgeai-gst-plugins/tree/fix/duplicate-frame-on-eos) and let us know if that solves the issue on your end.

aniket-l, 2022-01-10:
Yes this has resolved the issue.

closed 2022-01-10 by aniket-l