TexasInstruments/edgeai-gst-plugins #187: Video looping not working with tiovxmultiscaler

p-mantena, 2022-02-08:
Hi,

Not able to loop a video using tiovxmultiscaler. The video hangs when trying to seek.
Attaching a script file that reproduces the issue and output log and link for video input file.
Same pipeline works with videoscale.

[seek_sw_dec.txt](https://github.com/TexasInstruments/edgeai-gst-plugins/files/8021930/seek_sw_dec.txt)
[output.log](https://github.com/TexasInstruments/edgeai-gst-plugins/files/8021929/output.log)
[https://git.ti.com/cgit/edgeai/edge_ai_apps/](url)

Running with GST_DEBUG=2,
Error log :

APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=4) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
   821.559852 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
   821.559901 s:  VX_ZONE_INIT:Enabled
   821.559908 s:  VX_ZONE_ERROR:Enabled
   821.559914 s:  VX_ZONE_WARNING:Enabled
   821.560577 s:  VX_ZONE_INIT:[tivxInitLocal:130] Initialization Done !!!
   821.560817 s:  VX_ZONE_INIT:[tivxHostInitLocal:86] Initialization Done for HOST !!!
0:00:00.451183705  1713     0x3392be70 WARN                 basesrc gstbasesrc.c:3600:gst_base_src_start_complete:<filesrc0> pad not activated yet
0:00:00.451802170  1713     0x338b41e0 WARN                 qtdemux qtdemux.c:3250:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
0:00:00.505017570  1713     0x338b41e0 WARN            kmsallocator gstkmsallocator.c:549:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
0:00:00.581660610  1713     0x338b41e0 WARN            kmsallocator gstkmsallocator.c:549:gst_kms_allocator_dmabuf_import:<KMSMemory::allocator> Failed to close GEM handle: Invalid argument 22
0:00:05.457768375  1713     0x338b41e0 ERROR              tiovxsimo gsttiovxsimo.c:1360:gst_tiovx_simo_push_buffers:<tiovxmultiscaler0> Error pushing to pad: <tiovxmultiscaler0:src_0>
0:00:05.457837795  1713     0x338b41e0 ERROR              tiovxsimo gsttiovxsimo.c:1230:gst_tiovx_simo_chain:<tiovxmultiscaler0> Unable to push all buffers to source pads: -5
0:00:05.457885660  1713     0x338b41e0 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: Internal data stream error.
0:00:05.457900040  1713     0x338b41e0 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: streaming stopped, reason error (-5)
0:00:10.463534005  1713     0x338b41e0 ERROR              tiovxsimo gsttiovxsimo.c:1360:gst_tiovx_simo_push_buffers:<tiovxmultiscaler0> Error pushing to pad: <tiovxmultiscaler0:src_0>
0:00:10.463604855  1713     0x338b41e0 ERROR              tiovxsimo gsttiovxsimo.c:1230:gst_tiovx_simo_chain:<tiovxmultiscaler0> Unable to push all buffers to source pads: -5
0:00:10.463664115  1713     0x338b41e0 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: Internal data stream error.
0:00:10.463678745  1713     0x338b41e0 WARN                 qtdemux qtdemux.c:6619:gst_qtdemux_loop:<qtdemux0> error: streaming stopped, reason error (-5)
   831.977692 s:  VX_ZONE_INIT:[tivxHostDeInitLocal:100] De-Initialization Done for HOST !!!
   831.982095 s:  VX_ZONE_INIT:[tivxDeInitLocal:193] De-Initialization Done !!!
APP: Deinit ... !!!
REMOTE_SERVICE: Deinit ... !!!
REMOTE_SERVICE: Deinit ... Done !!!
IPC: Deinit ... !!!
IPC: DeInit ... Done !!!
MEM: Deinit ... !!!
MEM: Alloc's: 9 alloc's of 11061800 bytes 
MEM: Free's : 9 free's  of 11061800 bytes 
MEM: Open's : 0 allocs  of 0 bytes 
MEM: Deinit ... Done !!!
APP: Deinit ... Done !!!

Regards,
Prasanth Babu Mantena
.

m-herrera, 2022-02-15:
Hi @p-mantena 

We found that the multiscaler was not closing properly so after seeking, the pipeline was not transitioning to the PLAYING state properly. 
This is now fixed and merged to the develop branch; please give it a try on your end, and let us know if this issue can be closed.

Regards

p-mantena, 2022-02-24:
Hi @m-herrera 

Tested the develop branch. Looping is working fine now.
Thanks for the support.

Regards,
Prasanth

closed 2022-02-24 by p-mantena