TexasInstruments/edgeai-gst-plugins #193: Caps negotiation fails when a out buffer from tiovxmultiscaler is pushed to tiovxmosaic

ravi-rahul, 2022-02-21:
I have attached 2 simple scripts
[nv12_caps_colorconvert_py.txt](https://github.com/TexasInstruments/edgeai-gst-plugins/files/8108265/nv12_caps_colorconvert_py.txt)  - works fine
[nv12_caps_multiscaler_py.txt](https://github.com/TexasInstruments/edgeai-gst-plugins/files/8108266/nv12_caps_multiscaler_py.txt) - getting issue in tiovxmosaic caps negotioation

Below is the error log for the same"

_root@j7-evm:~# GST_DEBUG=tiovxmosaic:4 ./nv12_caps_multiscaler.py 
APP: Init ... !!!
MEM: Init ... !!!
MEM: Initialized DMA HEAP (fd=4) !!!
MEM: Init ... Done !!!
IPC: Init ... !!!
IPC: Init ... Done !!!
REMOTE_SERVICE: Init ... !!!
REMOTE_SERVICE: Init ... Done !!!
 10426.102037 s: GTC Frequency = 200 MHz
APP: Init ... Done !!!
 10426.102083 s:  VX_ZONE_INIT:Enabled

 10426.102089 s:  VX_ZONE_ERROR:Enabled
 10426.102094 s:  VX_ZONE_WARNING:Enabled
 10426.102518 s:  VX_ZONE_INIT:[tivxInitLocal:130] Initialization Done !!!
 10426.102691 s:  VX_ZONE_INIT:[tivxHostInitLocal:86] Initialization Done for HOST !!!
LAUNCHED
GOT SRC AND SINK
SRC PLAYED
SINK PLAYED
PULLING SAMPLE
PUSHING SAMPLE
PULLING SAMPLE
0:00:00.135865125  1588 0xffff7c020720 INFO             tiovxmosaic gsttiovxmosaic.c:931:gst_tiovx_mosaic_fixate_caps:<tiovxmosaic0> Fixating caps
0:00:00.135931340  1588 0xffff7c020720 ERROR            tiovxmosaic gsttiovxmosaic.c:956:gst_tiovx_mosaic_fixate_caps:<tiovxmosaic0> NULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLL

** (python3:1588): CRITICAL **: 09:56:29.522: gst_video_info_from_caps: assertion 'caps != NULL' failed

(python3:1588): GStreamer-CRITICAL **: 09:56:29.522: gst_mini_object_unref: assertion 'mini_object != NULL' failed
0:00:00.136110580  1588 0xffff7c020720 ERROR            tiovxmosaic gsttiovxmosaic.c:963:gst_tiovx_mosaic_fixate_caps:<tiovxmosaic0> Failed to get info from caps: (NULL)
PUSHING SAMPLE
PULLING SAMPLE
PUSHING SAMPLE_

m-herrera, 2022-03-03:
@ravi-rahul This is now fixed in develop; please give it a try and let us know if the issue is solved on your end.

ravi-rahul, 2022-03-04:
Hi @m-herrera 

Thanks! this issue is resolved

closed 2022-03-04 by ravi-rahul