TexasInstruments/edgeai-gst-plugins #221: New element for allocating GstBuffers from tiovx-memory

aniket-l, 2022-06-16:
Create a GStreamer element that can lend tiovx-buffers through DMABUF_IMPORT. It should allocate a bufferpool from the OVX shared DDR memory and pass it along to any element that uses DMABUF_IMPORT, for example v4l2h264dec.

ravi-rahul, 2022-06-19:
@m-herrera Can we use tiovxdelay for this purpose?

rrcarlosrodriguez, 2022-06-20:
@ravi-rahul , tivoxdelay can't be used because it doesn't allocate buffers. In any case, I would prefer creating a new element with only that purpose.

@aniket-l, we can work on this element, our estimate is 2 weeks to have it ready. Marco will start working on it today

m-herrera, 2022-06-29:
Hi @aniket-l, this has now been merged into develop, the documentation for the element can be found [here](https://github.com/TexasInstruments/edgeai-gst-plugins/wiki/tiovxmemalloc).

Feel free to test it and close this issue if it works as expected, otherwise please let us know so we can look into it.

aniket-l, 2022-07-13:
Tested

closed 2022-07-13 by aniket-l