TexasInstruments/edgeai-gst-plugins #225: Bug introduced in tiovx mosaic

ravi-rahul, 2022-07-04:
https://github.com/TexasInstruments/edgeai-gst-plugins/pull/220

with above PR a bug is introduced in tiovxmosaic
pos_x pos_y not working as expected

pipeline to reproduce the bug:

**_gst-launch-1.0 videotestsrc !  video/x-raw,format=NV12, width=1280, height=720 ! queue ! mosaic_0.sink_0 tiovxmosaic name=mosaic_0 sink_0::startx=320  sink_0::starty=180  sink_0::width=1280   sink_0::height=720 ! video/x-raw,format=NV12, width=1920, height=1080 ! kmssink sync=false driver-name=tidss_**

m-herrera, 2022-07-05:
Hi @ravi-rahul 

PR #220 introduced the functionality for the mosaic to behave as a demuxing element, where windows can be selected from any channel.
To achieve this the element had to change these properties to accept an array of values, so that each pad can configure its multiple possible channels. 

This is documented in the [mosaic wiki ](https://github.com/TexasInstruments/edgeai-gst-plugins/wiki/tiovxmosaic) where you can find some examples, you can also read about the properties format running `gst-inspect-1.0 tiovxmosaic` 

The pipeline you sent just needs to present these properties as a 1 element array. Like so:
 
```
gst-launch-1.0 videotestsrc ! video/x-raw,format=NV12, width=1280, height=720 ! queue ! mosaic_0.sink_0 \
tiovxmosaic name=mosaic_0 \
sink_0::startx="<320>" sink_0::starty="<180>" sink_0::width="<1280>" sink_0::height="<720>" ! \
video/x-raw,format=NV12, width=1920, height=1080 ! kmssink sync=false driver-name=tidss
```

Please give this a try and let us know if the issue persists.

ravi-rahul, 2022-07-05:
Hi Marco,

Thanks for the clarification !

closed 2022-07-05 by ravi-rahul