TexasInstruments/edgeai-gst-plugins #288: ninja build fails: cannot find dependencies

edtamarin, 2023-03-17:
I am trying to build the plugins on a system where I have the RTOS SDK installed (Ubuntu 18 LTS).

`meson build` command succeeds, yet when running `ninja build` I get the following errors:

```
In file included from ../gst-libs/gst/tiovx/gsttiovxmiso.c:68:
../gst-libs/gst/tiovx/gsttiovx.h:74:10: fatal error: edgeai_tiovx_target_kernels.h: No such file or directory
   74 | #include <edgeai_tiovx_target_kernels.h>

../gst-libs/gst/tiovx/gsttiovximagemeta.c: In function ‘gst_tiovx_image_meta_get_plane_info’:
../gst-libs/gst/tiovx/gsttiovximagemeta.c:126:24: error: ‘TIVX_IMAGE_IMAGEPATCH_ADDRESSING’ undeclared (first use in this function)
  126 |   vxQueryImage (image, TIVX_IMAGE_IMAGEPATCH_ADDRESSING, img_addr, sizeof (img_addr));
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../gst-libs/gst/tiovx/gsttiovximagemeta.c:126:24: note: each undeclared identifier is reported only once for each function it appears in
```

The build then fails. I am following the build instructions as outlined in the repository. This also seems like a recently introduced issue as earlier versions of the code could be compiled this way.

edtamarin, 2023-03-17:
Upon further inspection, the issue seems to be introduced after commit 1f1f3b480ccf62db1e753a089f2860465a033b3e.

ravi-rahul, 2023-04-13:
Hi @edtamarin 

You might be using older SDK 
Please migrate to latest SDK Release

https://www.ti.com/tool/download/PROCESSOR-SDK-LINUX-SK-TDA4VM

closed 2023-04-13 by ravi-rahul