TexasInstruments/edgeai-gst-plugins #347: Known Issue: ADASVISION-5113 - GST Pipeline error: Error pulling tensor from GST Pipeline

heebahsaleem, 2023-11-30:
Hi,

Is this known issue resolved?

I am trying to do this for the resolution 640X480 : [https://forum.beagleboard.org/t/csi-camera-not-working-on-beagleboard-ai-64-using-ti-model-zoo/36436/8  ](url)
but it is giving me error: 

[image]

any help would be appreciated

st311325, 2024-05-06:
"Hello, have you solved the problem above?"

st311325, 2024-05-06:
I have the same problem as you

darwinOne, 2024-07-30:
Hi @heebahsaleem @st311325 ,
I am also having this problem, have you guys found a solution?

st311325, 2024-07-30:
娇娇已收到

st311325, 2024-07-31:
修改推理视频的分辨率即可
 

娇娇的邮箱
***@***.***

darwinOne, 2024-07-31:
Can you explain it in more detail? @st311325 
I have changed the resolution, but the same error still occurs.
I encountered this issue when I ran rtsp_src_example.yaml on edge-gst-app with my RTSP source input.
Thank you!

st311325, 2024-08-01:
我更换了SD卡后重新部署也出现了这样的问题，如果找到解决办法会及时回复你

 

娇娇的邮箱
***@***.***

darwinOne, 2024-08-01:
Oke, Thank you!